tlambert03/LLSpy #5: Error : 'TiffPage' object has no attribute 'bits_per_sample'

hhakozaki, 2018-03-22:
When I specify folder which contain LLSM tiff data, v0.3.7 cause this error.

[image]

With executable distributed with Dropbox - v0.3.4 - won't cause this error and register folder fine.

BTW
 I really appreciate your effort to share your work.

Hiro.

tlambert03, 2018-03-22:
Hi Hiro, and thanks for the bug report.

That bug is caused by API change in an update in one of the dependencies (tifffile).  I didn't specify a tifffile version in my dependencies at first, so if you recently installed it, you got that new (broken) version.  (and the reason it works in the executable is that the older version of tifffile is bundled in it).

However, this problem is already be fixed in [version 0.3.8](https://github.com/tlambert03/LLSpy/releases/tag/0.3.8), which is available.  Just do `conda update llspy `  in your llsenv environment.

hhakozaki, 2018-03-23:
Hi Talley,
 Thank you for the information. I updated and confirmed it was fixed.

 It is great that your software utilize all available GPU. I have Tesla K20
and GTX 760 it worked great with these GPUs.

 Thank you for your work and sharing it with LLSM Community.

Hiro.

tlambert03, 2018-03-23:
Great!  Glad that fixed your problem, and glad you find it helpful.  Please don’t hesitate to report additional bugs as you find them (big or small).

closed 2018-03-23 by tlambert03