tlambert03/LLSpy #7: Reverse option has no effect

fruedinger, 2018-05-07:
Hi Talley,
it seems to me that the "reverse" option in rotation settings has no effect. My LLSpy will call cudaDeconv with the same settings and resulting images look the same.

Best,
Felix

tlambert03, 2018-05-07:
indeed!  I almost never use the rotate-to-coverslip feature and it looks like I never hooked that reverse option into cudaDeconv...  thanks for pointing that out. 
out of curiosity, do you have a system that does the stage scanning in the reverse direction?

fruedinger, 2018-05-07:
Yes I think so. The arrow on the sample scanning piezo points to the left, e.g. the sample will move to the left for positive voltages. I think this is because PI redesigned the piezo so the cable would hit the sample chamber if mounted as originally designed.

I compensated this by setting negative sample piezo intervals. Given that the deskewing in LLSpy worked fine for me I now realize that negative s steps are potentially also not recognized (at least they show positive in the processing tab)? So I guess I can actually work with the software as is!

tlambert03, 2019-02-14:
implemented in 0.4.0.

Felix, since I don't have a good dataset to test this with, let me know if you have a moment to test it out on your data and tell me if rotation is working well now.  If not, please reopen this issue, and send a small test dataset, with the settings.txt file included.

closed 2019-02-14 by tlambert03