tlambert03/LLSpy #9: Request: Compress final results as tiff

bscott711, 2018-06-26:
Hi Talley,

Would it be possible to add a compression option for the final results as tifs? I will typically use lzw compression, (compress=5 in tifffile) and get about a factor of 6 reduction in size, and see very minimal changes in time to load in either spimagine or imageJ after.

Thanks
-Brandon

tlambert03, 2018-06-28:
that should be possible.  I'm in the middle of rewriting llspy though and the new version will have much more flexible output options including hdf5, and also won't be using the cudaDeconv binary.  At the moment, to implement that change I would need to change way the CImg library is writing the tiff output in the cudadeconv binary itself... which is a bit more of a pain than just adding a parameter to a tifffile call (and would require compilation and redistribution of the new binary in the lattice dropbox).  Anyway: I agree it would be a useful feature and I'll work on getting it implemented asap.

tlambert03, 2019-02-14:
added directly to cudaDeconv in llspy_extra_0.4.0

closed 2019-02-14 by tlambert03