tlambert03/LLSpy #10: Deskewing: Set Background to Min/Mode Value

felixwaeldchen, 2018-07-10:
This is not an issue but rather a proposal.

I routinely use the deskewed stack for further analysis. With background set to 0, the moving window of actual data creates a hard edge that causes many issues for e.g. edge detecion or even single molecule localization.

It would be awesome to have an option to fill the background with a value other than 0. My suggestion would be to just use the value that can be selected in the preprocessing tab (fixed or mode value).

Best,
Felix

tlambert03, 2018-07-12:
i can probably do that pretty easily... will keep this open until added

tlambert03, 2019-02-14:
added manual pad value entry ... but haven't linked it to the preprocessing tab yet...

tlambert03, 2024-03-24:
closing as this project is in maintenance mode, likely will not add new features

closed 2024-03-24 by tlambert03