tlambert03/LLSpy #12: cudaDevonv binaries for platforms other than Windows

VolkerH, 2018-09-21:
Hi,
I'm trying to install llspy. I contacted janelia for a license and received a download link through flintbox. However, the dowload only contains a .zip file with executables and .dll files for Windows. Your installation notes refer to a file on Dropbox and looking at the source code the installer is looking for .so and .dylib files on Linux and Mac, respectively (as one would expect). Is there any chance to get the libraries for linux?

tlambert03, 2018-09-21:
Yeah, sounds like they gave you the wrong thing (the ones on flintbox won't even work with llspy on windows) ... did you specify that you needed the LLSpy libraries in the dropbox?  Or did you just ask for cudaDeconv?  Janelia's license prevent me from distributing the cudaDeconv libraries, so you will need to reply to them specifying what you need.  But I will also email them again to see if we can perhaps make it easier in the future.

VolkerH, 2018-09-21:
Hi, thanks for your reply.
In my email I had requested access to 

> cudaDeconv software written by Lin Shao and maintained by Dan Milkie

as specified in the Requirements section of the LLSpy `Readme.md`. I will send another request to innovatoin management asking specifically for the LLSpy libraries, hopefully whoever processes it will know what I'm referring to.

If you could get in touch with them and have your bundle of libraries referred to by a unique identifier, that would certainly be helpful for future users.

tlambert03, 2018-09-23:
yeah, that could definitely be clearer... sorry for the confusion.  I'll change it

tlambert03, 2018-09-23:
email me directly if Janelia doesn't respond within a day or so with an invite to the dropbox folder...

closed 2018-09-23 by tlambert03

VolkerH, 2018-10-05:
Hi Talley,

I will tack this on to this issue (despite the issue being closed) as it is related.
I meanwhile received an invite to a dropbox folder from Janelia. However, I'm not sure whether this is the latest folder.
I managed to run `lls install llspy_extra`:

[image]
 
When trying to run `llspy` I receive the following:

```
(llsenv) C:\Users\vhil0002\Desktop\LLSpy>lls -h
WARNING:llspy.libcudawrapper | Failed to properly import libcudaDeconv
function 'Deskew_interface' not found
```

With the GUI I get this additional dialog box:

[image]

This seems to indicate that maybe the dropbox folder link I was given is outdated?!
Below is the file size of the DLL, could you please compare it to the latest version 
and tell me whether this size is identical to the version that I need? If necessary I can also create 
an `md5`-checksum of the files:

```
(llsenv) C:\Users\vhil0002\Desktop\LLSpy\llspy_extra\win32\lib>dir
 Volume in drive C has no label.
 Volume Serial Number is 60E8-2FB3

 Directory of C:\Users\vhil0002\Desktop\LLSpy\llspy_extra\win32\lib

09/10/2017  07:57 PM    <DIR>          .
09/10/2017  07:57 PM    <DIR>          ..
09/10/2017  07:29 PM         1,734,144 libcudaDeconv.dll
```

And should I worry about the `Deskew_interface not found` message?

tlambert03, 2018-10-05:
Very sorry, please email me directly.