tlambert03/LLSpy #15: Minor Issues in v0.4

bscott711, 2019-02-14:
Hey Talley,

First off, great work on this! 
Following the update, the spimagine plugin works on my windows machine, so thank you!

I noticed three minor issues pop up.

1) Following rotation, the dz value is not being upsampled in the metadata, and is kept at the deskewed value, for a typical 500 nm stage step size, it is ~250 nm following deskewing, following rotation it should be ~100 nm. 

2) The metadata for the time interval in the MIPs was lost in this version, when loading into fiji it just has it listed as 0.

3) In ditch_partial_tiffs (llsdir.py line 725), the threshold of 1000 is problematic, because we have some images that are the same size but differ slightly in size for some weird reason even though they are straight off the camera. (164,756 KB vs 164,759 KB) . It may be better to use a threshold based on whether there is an image missing: thresh=ny*nx*bitDepth/8. For a 768x768 image that is 1179648

Thanks again,
-Brandon

tlambert03, 2019-02-14:
Great catches, thanks!

The metadata for rotation will require a change in cudaDeconv itself, and can be done and distributed as an update to llspy_extra. 
For the MIPs, you're talking about the combo-MIP generated by LLSpy right?  I'll figure out where that went...
Yeah, ditch_partial_tiffs is all-around problematic!  That was an early idea that I never returned to to improve. I like your idea and will implement something like that, in addition to, perhaps, just making the whole check optional.

bscott711, 2019-02-15:
Sounds good, and yes I was talking about the combo-mip. (Super helpful!)

There seems to be a major bottleneck in writing the file to disk following deconvolution.
It is taking ~1 minute to process a timepoint, and 56 seconds of that is writing the file to disk.
For a comparison, in version 0.3.9, the same data and settings takes 4 seconds, with 2 seconds of writing.

tlambert03, 2019-02-15:
oh no...  that's not good news at all.  What I can say is that multi-threaded IO got introduced into the cudaDeconv code, between the version of cudaDeconv that was used in 0.3.9 and the current one. 

Can you tell me what OS version and GPU you're using?  Also, does it matter whether you uncheck saveDeskewedRaw or uncheck all of the MIPs options?

 Pinging @linshaova and @dmilkie ...

bscott711, 2019-02-15:
[image]

I modified the llspy v.0.3.9 to make use of Lin's last cudaDeconv from ~Jul of last year, and that one is the comparison I am making. I don't have saveDeskewedRaw checked, I will test the MIPs option.

tlambert03, 2019-02-15:
Thanks.  Can you print the output from `cudaDeconv -h` for the version that works fine?

bscott711, 2019-02-15:
[image]

tlambert03, 2019-02-15:
just sent you a single-threaded version to test.  lemme know if it changes anything

dmilkie, 2019-02-15:
Here is the performance of the latest cudaDeconv that I compiled at the beginning of this month using the latest libraries (Feburary 2019): 
**9.58 seconds**
[image]

Here is the performance of the April 2017 version:
**10.54 seconds**
[image]

tlambert03, 2019-02-17:
Fixed it.  Something happened between CImg.h version 1.7.9 and version 2.0.0 that makes cudaDeconv hang on the last file for some reason (I was using 2.3.3, the regular cudaDeconv uses 1.5.8).  I wasn't able to figure out exactly what the problematic change was in CImg, but in any case, moving to 1.7.9 gives me all of the metadata interface I need, without the slowdown.  I'll update llspy_extra soon.

tlambert03, 2019-02-22:
issue 1 and the slowdown is fixed in llspy_extra_0.4.1.   issues 2 and 3 are fixed in 52c42c9 and bc4563e respectively.

closed 2019-02-22 by tlambert03