tlambert03/LLSpy #18: Flat field correction?

yemaopt, 2019-05-29:
Hi Talley!
I wonder if you can add the flat field correction easily into LLSpy, or the camera flash correction has already achieved the same function (even though I haven't tried that). Any other suggestions on improve the illumination uniformity in the hardware aspect are also appreciated. 
Thank you so much for your excellent work!
Sincerely yours,
Ye

tlambert03, 2019-05-31:
Hi Ye, sure I can add...  I've mostly been dragging my feet on that one cause it's another wavelength-specific setting (like OTF), and I've struggled to find an elegant way to incorporate it into the GUI.  It won't happen quickly though (busy at the moment) so If you want to try it out immediately just to see if it will help you, I'd recommend just using it directly in the cudaDeconv command line (using the --LSC flag).  On the hardware side, it has a lot to do with alignment of the laser at the SLM and lens choice (for instance, you could opt for a different cylindrical lens pair to spread out the gaussian a bit more at the SLM, at the expense of laser power/efficiency).   alternatively, you could dither the sheet over a larger extent (at the potential expense of sheet thickness, if your scan plane isn't perfectly coplanar with the detection objective).  I believe there was a conversation about this on the google group a while back?

tlambert03, 2024-03-24:
closing as this package is in maintenance mode.  no plans to add features

closed 2024-03-24 by tlambert03