tlambert03/LLSpy #19: Package conflicts on v0.4.3

bscott711, 2019-08-16:
Hi Talley,

I am trying a fresh install on a new workstation, but there are lots of conflicts in conda. Using the following command: conda create -n llsenv python=3.6 llspy
Here are the resulting conflicts I am getting:
[image]

tlambert03, 2019-08-16:
yeah, I just did a similar thing yesterday and also saw some conflicts...  but I think that was with python3.7.  In any case, when I definitely used 3.6 it did install and work ok on my machine (can't remember if it complained a bit at first).  Does it not install at all on yours?  
if not, what is your channel priority setting?

`$ conda config --show channel_priority`

bscott711, 2019-08-16:
It looks like it will install if I don't specify a python, but then it uses 2.7, if I specify with 3.6 or 3.7 it has lots of conflicts. 

The channel priority is flexible.

tlambert03, 2019-08-16:
Ok thanks, what platform?  I’ll check the dependencies this weekend

tlambert03, 2019-08-18:
I'm not sure if the [new environment solvers in conda 4.7](https://www.anaconda.com/how-we-made-conda-faster-4-7/) or disappearing upstream dependencies are to blame, but that really turned into a big pain in the ass.  I was eventually able to track down the conflict to some of the spimagine dependencies, and to pigz (which I used for compression on windows).

Neither of those packages are critical to llspy, so starting with LLSpy-0.4.3, I'm just not going to bundle those two packages with LLSpy on windows, so it will be up to the end user to install spimagine and/or some compatible compression binary (llspy will look for lbzip2, pbzip2, pigz, bzip2, gzip) if they want to use either of those features.  It may work to install llspy, activate the environment, and just `conda install -c talley spimagine`... but you may need to do some more hacking.

anyway, installing LLSpy (v0.4.3) should now work on windows and mac for both python 3.6 and (now added) 3.7 ... works for py3.6 on linux (still working on 3.7).  @bscott711, let me know if it solves your problem when you have a chance to try it.

bscott711, 2019-08-19:
Thanks Talley. The install works now (windows 10) without spimagine.

tlambert03, 2019-08-19:
OK, thanks for letting me know.

closed 2019-08-19 by tlambert03

bscott711, 2019-08-19:
I just wanted to follow up with dependencies I used to get spimagine working with the new version of llspy, if anyone came across it as well (windows 10)

conda create --name llsenv llspy spimagine=0.2.6 numpy==1.13.3 scipy==1.2.1 pyopengl==3.1.1a1 pyopengl-accelerate==3.1.1a1

tlambert03, 2019-08-19:
interesting...  When I installed spimagine with that pyopengl, I ended up getting errors upon launching spimagine, that looked very similar to what was described [here](https://github.com/winpython/winpython/issues/613).  **cgohlke** there mentioned it was a bug in pyopengl ... and the older versions didn't seem to be on anaconda cloud anymore.  So pyopengl seemed to be a bit of a wild-card there.  after installing, did you run "spimagine" at the command line and launch it successfully and everything,  or just from within llspy?