tlambert03/LLSpy #20: Fail to install spimagine on v0.4.3 and possibility to downgrade?

yemaopt, 2019-08-24:
Hi Talley!
I updated the LLSpy to v0.4.3 and found that spimagine is not installed automatically. I manually install it by using
conda install -c talley spimagine
and it will report the float128 error, which you mentioned in the previous post
[image]
I also tried Scott's suggestions by
conda config --add channels anaconda
conda create --name llsenv llspy spimagine=0.2.6 numpy==1.13.3 scipy==1.2.1 pyopengl==3.1.1a1 pyopengl-accelerate==3.1.1a1
But it results in "fail to import any qt binding" error and the lls gui cannot be ran
[image]
Could you give me some suggestions on this? And is it possible to downgrade to v0.4.2 where the spimagine is working, and currently I didn't use channel registration too much.
Many thanks for your help!
Sincerely yours,
Ye

tlambert03, 2019-08-25:
First off, it's possible to downgrade in one of two ways:

1. you can specify the version of llspy you'd like to install, just as you've done with the other packages.  so `conda install llspy=0.4.2` will install v0.4.2.
2. if you updated to llspy 0.4.3 from an existing conda environment (rather than creating a new environment), you can roll back using [conda revisions](https://sriramjaju.github.io/2018-05-30-2-minute-recipe-how-to-rollback-your-conda-environment/).  type `conda list --revisions` to show all the previous versions of your environment, then `conda install --revision <rev_num>` to rollback to a specific one.

I'm not sure why you wouldn't be getting qt bindings, since pyqt is a stated dependency of both llspy and spimagine... if you type `conda list qt`, what does it show installed in your environment?

For what it's worth, I really didn't change much about the way I package llspy for version 0.4.3, except to remove some dependencies... so the conflicts are coming somewhere lower down in the dependency chain.  There is a lot of platform-specific binary stuff for gputools/pyopencl/pyopengl ... so one possibility is that some of the specific versions that were working before (and which are specified in my recipes for gputools/spimagine or one of their dependencies)) has been removed from anaconda cloud.  If you are able to rollback to a specific revision that works for you, i'd be curious to see the exact versions of the full environment (with `conda list`).  

let me know if I can help further

Ducros, 2019-09-16:
I had a similar problem as Ye when I had to reinstall llspy after a computer crash.
I could not get spimagine to work on llsp0.4.3.
After a bit of trial and errors (and the halp of Talley and Ye)  I finally got both V0.4.2 and V0.4.3 working on my computer !
Here is the system I work on:
OS: Win 7 Prof. 64-bit
CPU: Intel Xeon E5-2680 V4 14-core
RAM: 128 GB
GPU: NVidia Quadro P6000 24GB RAM

An important detail (that I was not aware of) is the channels priority (see https://docs.conda.io/projects/conda/en/latest/user-guide/tasks/manage-channels.html). Basically channels "conda-forge" and "talley" have to be higher priority than "anaconda", otherwise packages are downloaded from "anaconda" channel and it does not work. However, you still need to have the "anaconda" channel otherwise a few packages are missing: e.g. pyopengl-accelarate3.1.1a1
If channels are in the right order of priority, the llspy0.4.2 fresh install on a new environment works fine for me:
conda create -n lls042 python=3.6 llspy=0.4.2

For llspy v0.4.3 I followed Talley s advices : I installed V0.4.3 and a new environment ("conda create -n llsenv python=3.6 llspy", then I used the command "conda install -c conda-forge -c talley spimagine=0.2.6 numpy=1.11.3 pyopengl=3.1.1a1 pyopengl-accelerate=3.1.1a1 pyqt=5.9.2" in the llsenv environment. It did not work at first, but then I replaced pyopencl by version 2017.2+cl12: "conda install pyopencl=2017.2+cl12".
I still have a numpy version warning message but both llspy0.4.3 and spimagine run OK.
I hope it helps other users.
Let me know if you want the list of packages on each environments too.
Mathieu

closed 2024-03-24 by tlambert03