tlambert03/LLSpy #21: LLSpy failed to process the stack of the multi FOV data after renaming

yemaopt, 2019-09-07:
Hi Talley!
I had a problem to process the data using the script function of the Labview software. I select 6 FOVs, and every 5min I take 1 volume stack for each FOVs, follwing the XY-T sequence. I rename the stacks using the LLSpy v0.4.2 provided function. Each FOV will be saved into a separate folder, and the preview looks good. But when I try to process them, for each FOV cudadeconv can only process 1 timepoint, and report illegal memory access error.
[image]
This only happens when processing the renamed dataset.
Do you have some ideas on this error? I will try to test the individual cudadeconv on this data set later.
Many thanks for your help!
Sincerely yours,
Ye

tlambert03, 2019-09-07:
I’ll need to see some example data, can you send me something via email/Dropbox?

yemaopt, 2019-09-07:
Hi Talley!
Here is the link.
https://www.dropbox.com/sh/5rtl27eekee7hzo/AADySLgEpBEWFvs8dlgoER3ra?dl=0
Many thanks!
Sincerely yours,
Ye

tlambert03, 2019-09-09:
Hey Ye,
I'm not really sure what's going on here, but I'm afraid this might just be a legitimate out-of-memory error.  These are rather large volumes after all.   I wasn't able to repeat the problem, with my Titan's (12 or 24GB VRAM).  If I try with a 1080 (8GB), it starts to crawl, due to RAM swapping... but I haven't seen it throw an illegal memory access yet. 

what GPU are you using and does the log state that it's resorting to "Use Host RAM..."?
are you running with with cudadeconv directly?  or via LLSpy?
does it happen whether or not you request rotation?  or only when rotating?
Is this the first time you've tried to process and rotate something as big as 1024x400x200?  (particularly after rotation and 32bit, that's going to make for a very big volume)

I don't think it has anything to do with file renaming...  it could be one of many things, but I'm pretty sure it has to do with something in cudaDeconv rather than llspy.  I realize it's weird that it doesn't seem to happen on the first timepoint, it's conceivable there's a memory leak, but I'm really not sure at the moment.

yemaopt, 2019-09-10:
Hi Talley!
Thank you so much for pointing this out! Yes, that's really the problem. I used GTX 1080 Ti GPU, and seems not be able to process the 1024 x 400 x 200 data. I was just wondering how the preview can work. Does LLspy downsample the raw data for preview calculation? When the preview is ongoing, it does mention use host RAM.
[image]
Thank you so much for your help!
Sincerely yours,
Ye

closed 2019-09-13 by yemaopt

tlambert03, 2019-09-13:
The major difference between the preview function and the “process” button is that the preview button doesn’t actually call cudadeconv, but rather directly calls the c functions from the shared library.  Similarly, the rotation there is done not using the cudadeconv binary, but with direct function calls.  I can’t remember off the top of my head, but it’s conceivable that the preview process is resetting the cuda device between the deconv and the rotation, perhaps making it work?  we might be able to  eek out a bit more memory efficiency when rotating in cudadeconv, but in the long run, if you’re going to be doing larger volumes and want to process with cuda, you’re probably going to want to invest in a gpu with larger vram.  The newer Titan RTX is pretty great bang-for-vram-buck

parthnatekar, 2022-05-16:
Hi, I'm getting this same error (albeit with a different error code).

I find that I can only process images below a certain size. I'm using a 24GB RTX 3090, so I doubt that I'm running out of memory.

[image]

linshaova, 2022-05-30:
Hi Parth, could you share the dataset and PSF/OTF used when this error occurred? Thanks!

lin