tlambert03/LLSpy #22: Fail to combine the MIP into one hyperstack for new exe version

yemaopt, 2019-10-13:
Hi Talley!
I got a new exe version for hardware acquisition and it works better in terms of timing, but when I process the data obtained by the exe, the MIP cannot be combined into a single hyperstack. This only happens for the data obtained using the new software. I guess this is related to the different format of the setting file?
Could you help me to check if this is solvable?
The data link is attached:
https://www.dropbox.com/s/mb78ffh7nddo1gy/multibead_sendtotalley.zip?dl=0
Thank you so much for your help!
Sincerely yours,
Ye

tlambert03, 2019-10-14:
Hi Ye,
I think it was related to the new placement of the "CAM1" in the filename, and a bad hard-coded filename pattern in the merge mips function.  I think [v0.4.4](https://github.com/tlambert03/LLSpy/releases/tag/0.4.4) should fix your issue... but it's definitely possible that you will run into more issue with expanded filenaming conventions of the new program.  so let me know if you do.
It's still uploading to anaconda, but you should be able to `conda update llspy` pretty soon and give it another shot.

closed 2019-10-14 by tlambert03

yemaopt, 2019-10-14:
Hi Talley!
Got it. Thank you so much! I would like to ask Dan first if "CAM1" can be
deleted, or I can write a simple matlab program to rename it (although
still more time consuming)
Many thanks for your help!
Sincerely yours,
Ye

tlambert03, 2019-10-14:
nah, that's not something that should need to be dealt with on Dan's side.  LLSpy _should_ be able to handle a wide variety of filenaming patterns... and it sort of can with the more recent "filename pattern" setting in the config tab... but there are still places in the code straggling behind that have issues.

Truth is, LLSpy was written with a relatively narrow set of conditions in mind.  I'm working on a more flexible rewrite, but for now I'll just try to patch these little problems in the current version as they come along.  Thanks!

yemaopt, 2019-10-14:
Hi Talley!
Got it. Thank you always for maintaining this excellent LLspy!
Sincerely yours,
Ye

yemaopt, 2019-10-14:
Hi Talley!
Could I ask an unrelated question here? How fast should I expect the
spimagine to play the video. For me, it usually takes 8s to load a 1138 by
1024 by 467 pixel volume of 106MB, although the demo video of spimagine
shows a very fast video playing on his demo data. I wonder if there is
better way to play the volume video more quickly.
Thank you so much for your help!
Sincerely yours,
Ye

tlambert03, 2019-10-14:
8 seconds starting when and ending when? is this from the moment you press "preview" to when the window appears?  or something else?

yemaopt, 2019-10-14:
Hi Talley!
It takes 8 second for spimagine to switch from one frame to another frame,
so to see the whole movie takes long. I wonder if that is normal and we
have better software to process that. I also tried Imaris, and it seems
similar.
Many thanks!
Sincerely yours,
Ye

tlambert03, 2019-10-14:
hmm, that does seem rather slow.  with the bead data you sent, on my system (w/ GTX 1080 Ti), it takes about 6-10 seconds per timepoint to render the first image (decon/rotation time), but then once spimagine is open, it renders movies at around 3 fps.
Note though: that data, once deskewed, even at 16 bits, is more like 1.06 GB... rather than 106MB.  So it's probably asking a lot to go much faster than a couple frames per second, unless you resample it into some pyramidal scheme like HDF5, as imaris does.  But even at that size, 8 seconds to render a frame seems slow.  I'm not entirely what would be causing that on your system.
I would try to take LLSpy out of the equation... just launch spimagine directly from the command line in your llsenv environment (type "spimagine" and a window should open).  Then drag, for instance,  a GPUdecon folder (without MIPS) into the window.  If it's still slow there, I would consider posting something to the issue tracker on the spimagine page.  If it's only slow within LLSpy, let me know.

yemaopt, 2019-10-14:
Hi Talley!
Many thanks for your information! I launched the spimagine solely and it
doesn't make difference. I just feel like the GPU is not working when I run
the spimagine. Could you tell me what OS you are using? Maybe I need to try
Linux instead. I will consider posting an issue to spimagine.
Thank you so much for your help!
Sincerely yours,
Ye

tlambert03, 2019-10-15:
I'm on OS X at the moment... 
You might try to create a config file for spimagine and/or gputools to force it to use the right device (if it isn't currently), as described here: https://github.com/maweigert/spimagine#configuration

you can also try this from your anaconda prompt command line, to see what device is being used.

```bash
python -c "import gputools; gputools.get_device().print_info()"
```

commit 22cc402f (2019-10-14): fixes #22