tlambert03/LLSpy #30: GPU cannot be recognized

yemaopt, 2020-04-19:
Hi Talley!
Sorry that I currently suffer the problem that the GPU I used cannot be recognized by CUDA, shown as attached 
[image]
Before I didn't have this problem, until I installed the cudatoolkit 9.1 and MS2015 for other purpose on matlab codes. I reinstall the llspy and it doesn't help. 
Could you give me some hint how I can solve it? I guess the cudatool kit version can be an issue.
[image]

Thank you so much for your help!

tlambert03, 2020-04-30:
does running cudadeconv from the command line give you any useful errors?

yemaopt, 2020-05-02:
Hi Talley!
Thank you so much for your hint! Yes, it shows the driver version is older, and thus I updated the NVIDIA driver, and recompile the MEX for the other matlab code, so that both work well finally.
Many thanks for your help!
Ye

closed 2020-05-02 by yemaopt

tlambert03, 2020-05-02:
great!  glad you were able to figure it out!