tlambert03/LLSpy #33: FiducialReg saving matrix

joaomamede, 2021-07-19:
What is the most appropriate and efficient way to save a bead registration to use at anytime with other experiment (basically to just to the conversion matrix once and apply subsequently).

Thanks!

tlambert03, 2021-07-19:
It's a good question, and I've been wanting to use just the figucialreg part of this package myself recently (without having to use LLSpy).  Truth is, it's been a long time since I've looked at this module, and I need to remind myself how it works :joy: ...

To be honest, I also feel quite certain that there are probably better packages that are more worth your time.  I assume that [ITK](https://github.com/InsightSoftwareConsortium/ITK) would be the place to go for this kind of thing.

@thewtex: If I may embarrassingly _not_ rtfd for just a moment ... can you point me/us to the function(s) in ITK that we should read about here?  The goal is to estimate the affine matrix that maps one 3D image of a calibration image set of fluorescent beads/point sources to another (so that it can then be applied to experimental data collected later).  This package used coherent point drift (which seemed to work well for that type of data), but what would you recommend these days in the python world, and does ITK have a good routine for this?

joaomamede, 2021-07-19:
The good thing about your FigReg, is that it's gpu speed btw! (My files are ...sometimes 1.5TB). I'll look into ITK.

tlambert03, 2021-07-19:
ah, true!  well mostly it's just the actual affine transform that's gpu-accelerated (not the calculation of the transform).  and just so that you're aware, the "best" parts of LLSpy (namely, the decon and affine transforms) have been extracted to pycudadecon, which recently got a lot easier to install since the underlying cuda code is now on conda-forge.  So check out https://github.com/tlambert03/pycudadecon and the [affine module](https://www.talleylambert.com/pycudadecon/affine.html) if you're looking for that part ... (and of course, there's [cupy's affine transforms](https://docs.cupy.dev/en/stable/reference/generated/cupyx.scipy.ndimage.affine_transform.html) as well)

joaomamede, 2021-07-20:
I used pycudadecon before. pretty good. I actually have an issue with it I should report.

Thanks for all your coding Talley.

joaomamede, 2021-08-06:
Is it a direct usage with scipy/cupy affine_transform? As in the 'matrix' argument in cupy?

closed 2024-03-24 by tlambert03