tlambert03/LLSpy #34: Installation Errors, glibc incompatibility

parthnatekar, 2021-12-02:
Hi,

I am unable to install using conda.

```
UnsatisfiableError: The following specifications were found to be incompatible with each other:

Output in format: Requested package -> Available versions
The following specifications were found to be incompatible with your system:

  - feature:/linux-64::__glibc==2.27=0
  - feature:|@/linux-64::__glibc==2.27=0

Your installed version is: 2.27
```

Not sure how to solve this glibc error. I'm creating a new Python3.6 environment so there should be no other conflicts.

tlambert03, 2021-12-02:
Can you tell me what OS you're trying to install on?

parthnatekar, 2021-12-02:
Its Ubuntu 18.04

parthnatekar, 2022-01-05:
Hi, any updates on this, still getting the same error!

tlambert03, 2022-01-05:
I actually did do some updates in early december... but not sure if it's going to fix this.  Can you try this exactly and let me know what happens?

```sh
conda create -n newenv -c conda-forge python=3.9 cudadecon
conda activate newenv
pip install llspy[pyqt,napari]
lls gui
```

parthnatekar, 2022-01-25:
Hi,

Thanks for your updates.

I was able to install and import llspy using the above. I'm closing this for now but I'll let you know if I have any trouble later on.

Thanks

parthnatekar, 2022-01-25:
Hi, getting the following error now on running ```process``` or ```autoprocess```:

```
llspy.exceptions.CUDAbinException: cudadecon could not be located or is not executable: /home/schoeneberglab/miniconda3/envs/newenv/bin/cudadecon

```

tlambert03, 2022-01-25:
output of `conda list` oplease?

parthnatekar, 2022-02-08:
```
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                       1_gnu    conda-forge
alabaster                 0.7.12                   pypi_0    pypi
appdirs                   1.4.4                    pypi_0    pypi
asttokens                 2.0.5                    pypi_0    pypi
attrs                     21.4.0                   pypi_0    pypi
babel                     2.9.1                    pypi_0    pypi
backcall                  0.2.0                    pypi_0    pypi
black                     21.12b0                  pypi_0    pypi
boost-cpp                 1.74.0               h359cf19_5    conda-forge
bzip2                     1.0.8                h7f98852_4    conda-forge
ca-certificates           2021.10.8            ha878542_0    conda-forge
cachey                    0.2.1                    pypi_0    pypi
certifi                   2021.10.8                pypi_0    pypi
charset-normalizer        2.0.10                   pypi_0    pypi
click                     8.0.3                    pypi_0    pypi
cloudpickle               2.0.0                    pypi_0    pypi
cudadecon                 0.0.11               h32a6c52_0    conda-forge
cudatoolkit               11.6.0               habf752d_9    conda-forge
cycler                    0.11.0                   pypi_0    pypi
dask                      2022.1.0                 pypi_0    pypi
debugpy                   1.5.1                    pypi_0    pypi
decorator                 5.1.1                    pypi_0    pypi
docstring-parser          0.13                     pypi_0    pypi
docutils                  0.17.1                   pypi_0    pypi
entrypoints               0.3                      pypi_0    pypi
executing                 0.8.2                    pypi_0    pypi
fftw                      3.3.10          nompi_h77c792f_102    conda-forge
fonttools                 4.29.0                   pypi_0    pypi
freetype-py               2.2.0                    pypi_0    pypi
fsspec                    2022.1.0                 pypi_0    pypi
heapdict                  1.0.1                    pypi_0    pypi
hsluv                     5.0.2                    pypi_0    pypi
icu                       69.1                 h9c3ff4c_0    conda-forge
idna                      3.3                      pypi_0    pypi
imageio                   2.14.1                   pypi_0    pypi
imagesize                 1.3.0                    pypi_0    pypi
importlib-metadata        4.10.1                   pypi_0    pypi
intervaltree              3.1.0                    pypi_0    pypi
ipykernel                 6.7.0                    pypi_0    pypi
ipython                   8.0.1                    pypi_0    pypi
ipython-genutils          0.2.0                    pypi_0    pypi
jbig                      2.1               h7f98852_2003    conda-forge
jedi                      0.18.1                   pypi_0    pypi
jinja2                    3.0.3                    pypi_0    pypi
jpeg                      9d                   h36c2ea0_0    conda-forge
jsonschema                4.4.0                    pypi_0    pypi
jupyter-client            7.1.2                    pypi_0    pypi
jupyter-core              4.9.1                    pypi_0    pypi
kiwisolver                1.3.2                    pypi_0    pypi
ld_impl_linux-64          2.36.1               hea4e1c9_2    conda-forge
lerc                      3.0                  h9c3ff4c_0    conda-forge
libdeflate                1.8                  h7f98852_0    conda-forge
libffi                    3.4.2                h7f98852_5    conda-forge
libgcc-ng                 11.2.0              h1d223b6_12    conda-forge
libgfortran-ng            11.2.0              h69a702a_12    conda-forge
libgfortran5              11.2.0              h5c6108e_12    conda-forge
libgomp                   11.2.0              h1d223b6_12    conda-forge
libnsl                    2.0.0                h7f98852_0    conda-forge
libstdcxx-ng              11.2.0              he4da1e4_12    conda-forge
libtiff                   4.3.0                h6f004c6_2    conda-forge
libuuid                   2.32.1            h7f98852_1000    conda-forge
libwebp-base              1.2.2                h7f98852_1    conda-forge
libzlib                   1.2.11            h36c2ea0_1013    conda-forge
llspy                     0.5.0                    pypi_0    pypi
llspy-slm                 0.2.1                    pypi_0    pypi
llvmlite                  0.38.0                   pypi_0    pypi
locket                    0.2.1                    pypi_0    pypi
[... 66 lines cut ...]
sqlite                    3.37.0               h9cd32fc_0    conda-forge
stack-data                0.1.4                    pypi_0    pypi
superqt                   0.2.5.post1              pypi_0    pypi
tifffile                  2021.11.2                pypi_0    pypi
tk                        8.6.11               h27826a3_1    conda-forge
tomli                     1.2.3                    pypi_0    pypi
toolz                     0.11.2                   pypi_0    pypi
tornado                   6.1                      pypi_0    pypi
tqdm                      4.62.3                   pypi_0    pypi
traitlets                 5.1.1                    pypi_0    pypi
typer                     0.4.0                    pypi_0    pypi
[... cut ...]
watchdog                  2.1.6                    pypi_0    pypi
wcwidth                   0.2.5                    pypi_0    pypi
wheel                     0.37.1             pyhd8ed1ab_0    conda-forge
wrapt                     1.13.3                   pypi_0    pypi
xz                        5.2.5                h516909a_1    conda-forge
zipp                      3.7.0                    pypi_0    pypi
zlib                      1.2.11            h36c2ea0_1013    conda-forge
zstd                      1.5.2                ha95c52a_0    conda-forge
```

parthnatekar, 2022-04-20:
Hi, any updates on this? I believe it is just a question of pointing LLSPY to the cudaDecon installation.

tlambert03, 2022-04-20:
> I believe it is just a question of pointing LLSPY to the cudaDecon installation.

this is what the purpose of llspy and https://github.com/tlambert03/LLSpy/pull/35 and https://github.com/tlambert03/LLSpy/pull/36 ... so this seems to suggest a different problem.

So, if you run `cudadecon` on the command line, it is installed right?  I'm guessing there were more places that needed to be swapped over.  This has actually gotten a bit harder for me to test, since my lattice and llspy rig isn't running at the moment.  If you'd like to help, could you try something for me?

In your llspy installation, go in to the `process` function here:

https://github.com/tlambert03/LLSpy/blob/39b344e8eaa4b7e02fb29a363bb26c0c1905874e/llspy/llsdir.py#L1139-L1146

and print out the value for `binary` (both before and after the check for "if is None") ... I want to see if something (like a configuration somewhere) is providing something other than None

parthnatekar, 2022-04-22:
Hi, 

Yes, I can run cudaDecon from the command line.

When I run ```E.autoprocess()```, an error occurs at ```line 418 in llsdir.py```

```
if binary is None:
        binary = CUDAbin()
```

If I print out ```binary``` above this I get ```None```.

tlambert03, 2022-04-23:
and if you print it out afterwards?  (`print(binary.path)`) ...

parthnatekar, 2022-04-25:
It fails at that line, so I can't print out anything afterward. The entire error stack is:

```
Traceback (most recent call last):
  File "<<<PATH>>>", line 6, in <module>
    E.autoprocess(otfDir = '/media/schoeneberglab/DATA 01/Parth/deskew_decon/20190208/OTF')
  File <<<PATH>>>/newenv/lib/python3.9/site-packages/llspy/llsdir.py", line 1106, in autoprocess
    return process(self, **kwargs)
  File "<<<PATH>>>/newenv/lib/python3.9/site-packages/llspy/llsdir.py", line 418, in process
    binary = CUDAbin()
  File "<<<PATH>>>/newenv/lib/python3.9/site-packages/llspy/cudabinwrapper.py", line 215, in __init__
    binPath = get_bundled_binary()
  File "<<<PATH>>>/newenv/lib/python3.9/site-packages/llspy/cudabinwrapper.py", line 101, in get_bundled_binary
    raise CUDAbinException(
llspy.exceptions.CUDAbinException: cudadecon could not be located or is not executable: <<<PATH>>>/newenv/bin/cudadecon
```

tlambert03, 2022-04-25:
ok, can you do me one more favor?  I'm wondering if my (pretty old) logic for determining if a given program name is executable in the current environment is not working for you.

can you edit `get_bundled_binary` in `cudabinwrapper.py` [here](https://github.com/tlambert03/LLSpy/blob/main/llspy/cudabinwrapper.py#L72) to 
A) check that it is indeed trying to find the program `cudadecon` (which you said _does_ work fine when run directly).  i,e, print the value of `name` right in the beggining of the function.
B) if it is indeed trying to find `cudadecon`, see if you can figure out why `util.which` (which is called [right here](https://github.com/tlambert03/LLSpy/blob/main/llspy/cudabinwrapper.py#L100) and [defined here](https://github.com/tlambert03/LLSpy/blob/39b344e8eaa4b7e02fb29a363bb26c0c1905874e/llspy/util.py#L111-L149)) isn't working?

parthnatekar, 2022-04-25:
So all I had to do was change the default value for the ```name``` argument from ```cudadecon``` to ```cudaDecon``` and it was able to find what was needed.

Seems to be working now. I'm not sure if this was something to do with how different systems handle upper case and lower case paths.

Thanks for your support, I'll keep this open for a few more days if you don't mind in case any more problems come up.