tlambert03/LLSpy #42: lls gui

BCinquin, 2023-12-05:
* {{ cookiecutter.project_name }} version:
* Python version: Python 3.6.15
* Operating System: Windows 10

### Description

Starting the GUI

nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2023 NVIDIA Corporation
Built on Fri_Nov__3_17:51:05_Pacific_Daylight_Time_2023
Cuda compilation tools, release 12.3, V12.3.103
Build cuda_12.3.r12.3/compiler.33492891_0

### What I Did

```
After following the instruction to install LLSPY.

(llsenv) C:\Users\Bertrand>lls gui
Traceback (most recent call last):
  File "C:\Users\Bertrand\Anaconda\envs\llsenv\Scripts\lls-script.py", line 6, in <module>
    from llspy.bin.llspy_cli import cli
  File "C:\Users\Bertrand\Anaconda\envs\llsenv\lib\site-packages\llspy\__init__.py", line 21, in <module>
    from .llsdir import LLSdir, preview, process, RegDir
  File "C:\Users\Bertrand\Anaconda\envs\llsenv\lib\site-packages\llspy\llsdir.py", line 22, in <module>
    from . import parse, schema, util
  File "C:\Users\Bertrand\Anaconda\envs\llsenv\lib\site-packages\llspy\schema.py", line 3, in <module>
    from voluptuous import (
  File "C:\Users\Bertrand\Anaconda\envs\llsenv\lib\site-packages\voluptuous\__init__.py", line 3, in <module>
    from voluptuous.schema_builder import *
  File "C:\Users\Bertrand\Anaconda\envs\llsenv\lib\site-packages\voluptuous\schema_builder.py", line 1
    from __future__ import annotations
    ^
SyntaxError: future feature annotations is not defined
```
Thank you for your help...

tlambert03, 2023-12-05:
hi.  At this point, this package is largely in maintenance mode, so while we can probably figure this particular problem out for you, I'd encourage you to start considering alternatives for your workflow.  maybe @pr4deepr can comment on whether https://github.com/BioimageAnalysisCoreWEHI/napari_lattice can replace llspy at this point?  ... or perhaps also suggest other things to consider?

anyway, to your questions specifically: it looks like whatever version of `voluptuous` you ended up with has dropped support for python 3.6... so either you need to update your version of python (which, i'm not sure, may break stuff about llspy), or you need to downgrade various packages in your environment to be compatible with python 3.6.  The first thing you could try is `pip install 'voluptuous<0.13'` and see if that works?

BCinquin, 2023-12-06:
> Thanks for pinging me @tlambert03 <https://github.com/tlambert03>. Lot of
> napari_lattice is inspired by LLSpy and your code, so thanks Talley!
>
> I've included support for deskewing in X and Y in napari_lattice, so it
> should work. I have tested it on data from the Janelia lattice design
> successfully.
> However, it does not read the folder of tiffs or the text file with
> metadata. You may have to resave it as a tiff stack or load the folder of
> tiffs in napari. Otherwise, you can enter all the other metadata.
>
> We're in the middle of making some changes
> <https://github.com/BioimageAnalysisCoreWEHI/napari_lattice/pull/47>; its
> taking much longer than we thought and has grown bigger! :)
>
> Cheers
> Pradeep

Thank you very much !

Napari_Lattice it is though !

Best regards

Bertrand

closed 2024-03-24 by tlambert03