tlambert03/LLSpy #45: can llspy package allow python version >=3.8?

linshaova, 2024-03-15:
* {{ llspy }} version: 0.4.9
* Python version: 3.6, 3.7
* Operating System: Windows 10

### Description

I haven't used `fiducialreg` for years. When I tried it recently and possibly after many updates of the underlying packages, there's now an error message when calling the `blur` function (see below of the detailed traceback). It seems to come down to `metadata` not available in `importlib`, which probably means Python 3.8 or above is needed for one of the packages?

### What I Did
I tried to reinstall `llspy` with `python=3.8` specified but found out there is an upper limit of <3.8 so this didn't go through. Could this upper limit be relaxed? Thanks!

Traceback is here:
```
Traceback (most recent call last):
  File "C:\Users\ls977\Miniconda3\envs\llsenv_py37\lib\site-packages\pyopencl\_mymako.py", line 2, in <module>
    import mako.template  # noqa
  File "C:\Users\ls977\Miniconda3\envs\llsenv_py37\lib\site-packages\mako\template.py", line 19, in <module>
    from mako import cache
  File "C:\Users\ls977\Miniconda3\envs\llsenv_py37\lib\site-packages\mako\cache.py", line 7, in <module>
    from mako import util
  File "C:\Users\ls977\Miniconda3\envs\llsenv_py37\lib\site-packages\mako\util.py", line 14, in <module>
    from .compat import importlib_metadata_get
  File "C:\Users\ls977\Miniconda3\envs\llsenv_py37\lib\site-packages\mako\compat.py", line 8, in <module>
    from importlib import metadata as importlib_metadata
ImportError: cannot import name 'metadata' from 'importlib' (C:\Users\ls977\Miniconda3\envs\llsenv_py37\lib\importlib\__init__.py)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "RegCalib.py", line 36, in <module>
    cloudset=fiducialreg.CloudSet(calib_list, labels=ims.wavelengths, dx=ims.dxy, dz=ims.dz, mincount=50)
  File "C:\Users\ls977\Miniconda3\envs\llsenv_py37\lib\site-packages\fiducialreg\fiducialreg.py", line 730, in __init__
    **kwargs
  File "C:\Users\ls977\Miniconda3\envs\llsenv_py37\lib\site-packages\fiducialreg\fiducialreg.py", line 546, in __init__
    self.update_coords()
  File "C:\Users\ls977\Miniconda3\envs\llsenv_py37\lib\site-packages\fiducialreg\fiducialreg.py", line 591, in update_coords
    if self.filtered is None:
  File "C:\Users\ls977\Miniconda3\envs\llsenv_py37\lib\site-packages\fiducialreg\fiducialreg.py", line 321, in __get__
    value = self.func(instance)
  File "C:\Users\ls977\Miniconda3\envs\llsenv_py37\lib\site-packages\fiducialreg\fiducialreg.py", line 574, in filtered
    from gputools import blur
  File "C:\Users\ls977\Miniconda3\envs\llsenv_py37\lib\site-packages\gputools\__init__.py", line 18, in <module>
    from gputools.core.oclalgos import OCLReductionKernel, OCLElementwiseKernel
  File "C:\Users\ls977\Miniconda3\envs\llsenv_py37\lib\site-packages\gputools\core\oclalgos.py", line 5, in <module>
    from pyopencl import elementwise, reduction, scan, algorithm
  File "C:\Users\ls977\Miniconda3\envs\llsenv_py37\lib\site-packages\pyopencl\scan.py", line 39, in <module>
    import pyopencl._mymako as mako
  File "C:\Users\ls977\Miniconda3\envs\llsenv_py37\lib\site-packages\pyopencl\_mymako.py", line 5, in <module>
    "Some of PyOpenCL's facilities require the Mako templating engine.\n"
ImportError: Some of PyOpenCL's facilities require the Mako templating engine.
You or a piece of software you have used has tried to call such a
part of PyOpenCL, but there was a problem importing Mako.

You may install mako now by typing one of:
- easy_install Mako
- pip install Mako
- aptitude install python-mako

or whatever else is appropriate for your system.

tlambert03, 2024-03-15:
yeah, was working on it in https://github.com/tlambert03/LLSpy/pull/43/files
but it fell by the wayside since other things were getting more usage/activity.  I'll try to pick it up again and get it across the finish line

tlambert03, 2024-03-24:
hey @linshaova, could you do me a huge favor and test out the main branch for me?  You can use anything up to python 3.11 (3.12 might work but haven't tested).  Please try as per the readme, subbing in the github url for the install step.  If everything imports ok for you and basic functionality works, I'll push a new release to PyPI.  thanks!

```sh
conda create -n llsenv python=3.11 cudadecon
conda activate llsenv
pip install git+https://github.com/tlambert03/LLSpy.git
```

linshaova, 2024-03-25:
Hi Talley, thanks for your hard work! I was able to install llspy with python=3.11. When I tried to run `fiducialreg` however, I got a very strange error (see below traceback). It looks like it got past the line in `update_coords()` where the issue occurred last time. However, somehow `self.autothresh()`  at line 590 is called (which throws the error below) even though that line shouldn't have been reached because `thresh == None` as far as I can tell (`update_coords()` invoked at line 541). 

Any clue?

```
Warning: Could not find scikit-tensor which is needed for separable approximations...
If you want to compute separable approximations, please install it with
pip install scikit-tensor-py3
Traceback (most recent call last):
  File "F:\OneDriveSync\OneDrive - Yale University\DeCamilli Lab\Registration\RegCalib.py", line 37, in <module>
    cloudset=fiducialreg.CloudSet(calib_list, labels=ims.wavelengths, dx=ims.dxy, dz=ims.dz, mincount=50)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\ls977\Miniconda3\envs\llsenv_py311\Lib\site-packages\fiducialreg\fiducialreg.py", line 713, in __init__
    FiducialCloud(
  File "C:\Users\ls977\Miniconda3\envs\llsenv_py311\Lib\site-packages\fiducialreg\fiducialreg.py", line 541, in __init__
    self.update_coords()
  File "C:\Users\ls977\Miniconda3\envs\llsenv_py311\Lib\site-packages\fiducialreg\fiducialreg.py", line 590, in update_coords
    thresh = self.autothresh()
             ^^^^^^^^^^^^^^^^^
  File "C:\Users\ls977\Miniconda3\envs\llsenv_py311\Lib\site-packages\fiducialreg\fiducialreg.py", line 582, in autothresh
    return get_thresh(self.filtered, mincount)[0]
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\ls977\Miniconda3\envs\llsenv_py311\Lib\site-packages\fiducialreg\fiducialreg.py", line 101, in get_thresh
    modecount = stats.mode(object_count[(object_count >= mincount)], axis=None).mode[0]
                ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^
IndexError: invalid index to scalar variable.
```

tlambert03, 2024-03-25:
please give it another try after running `pip install git+https://github.com/tlambert03/LLSpy.git` again