tlambert03/LLSpy #46: "point cloud" definition in fiducialreg

linshaova, 2024-03-26:
Hi Talley,

Would a calibration 4-channel image like the following work for `fiducialreg`? In other words, how strict is the "point cloud" requirement? The calibration results I got from this dataset seem to not work very well, which made me think maybe the sample is up to standard.
thanks!

-lin

[image]

tlambert03, 2024-03-26:
hey @linshaova, it "should" work...  but I'm sure there's a limit
The point cloud registration algorithm is as in https://arxiv.org/abs/0905.2635, which is designed to be robust to missing data points in one or more of the datasets (i.e. you don't need a 1-to-1 match of detected points to get a good result).  

It's a bit hard to tell from that image, i think what we really need to see are the pre-registered point clouds extracted from the raw data (based on bead detection).  It's been a while since I've looked at this code, would you be able to put together a brief script showing showing how you're using it, along with the dataset somewhere?

linshaova, 2024-03-26:
I didn't mean to ask you to debug this for me! Great reminder that `CloudSet` can be displayed and checked. It looks like if I use the `mincount` option, the number of found points varies greatly between channels. And strangely, I saw when one channel has >800 points and the other one has <50, `count_matching` somehow equals to almost 600! I had thought `count_matching` should be less than the fewer of the two channels?

In any case, I'm going to try specifying `threshold` when calling `CloudSet` and see if it improves anything.

tlambert03, 2024-03-26:
yeah i do vaguely recall that visualizing the resulting cloudsets was important...

linshaova, 2024-03-28:
After re-calibrate with manually setting `threshold`, instead of using `mincount`, in calling `CloudSet()`, the results look much better! Thanks!

tlambert03, 2024-03-28:
great!  yeah that aligns with my memory as well.  if you hit on more robust ways to auto-pick the threshold, i'm sure we could do better there

closed 2024-03-28 by tlambert03